mcous/gerber-parser #3: Support excellon slot command G85

katyo, 2015-10-21:
I think Gerber/Excellon parser is amazing thing for our evil js-world. ^^ Thanks for this great work, which replaced my old parser. But I have some problems.

I assume that slot drilling command G85 is one of important things, which must be supported by parser.

For example,
```
X154.061Y-90.652G85X153.747Y-91.618
```

mcous, 2015-10-21:
Glad to hear you like it!

I haven't ever encountered a G85 before, any chance you have a link to some documentation so I can code this up? I'm assuming it's formatted like:

`X[START_X]Y[START_Y]G85X[END_X]Y[END_Y]`

Where this command creates a slot with round ends from `(START_X, START_Y)` to `(END_X, END_Y)`

katyo, 2015-10-21:
Yes, this is that you think.
I found it in excellon specification:

> Slot
>
> G85
>
> G85 Cuts a slot by drilling a series of closely spaced holes between two points. The start of the hole is programmed with an X and Y coordinate, followed by the command, followed by the ending X and Y coordinate. The tool is specified with a T command prior to the G85 command. The tool size MUST be specified prior to using this command. The size may be provided by the Operator through the console, in the part program body, or the part program (M48) header. The slot is as wide as the drill bit used. The slot is created by drilling a series of evenly spaced adjacent holes from one end of the slot to the other. This leaves protrusions around the edge of the hole. Then another set of spaced holes is drilled between the previous set. This continues until a smooth sided slot has been produced. The holes are spaced close enough that these protrusions are less than 0.0005". [#1]

[#1]: http://web.archive.org/web/20071030075236/http://www.excellon.com/manuals/program.htm

mcous, 2015-10-21:
Sweet. I haven't been able to load the Excellon spec webpage for a while; I think they must've taken the it down. Even the Wayback machine isn't working for me.

I think this should be handled by emitting a `{cmd: 'set', line: LINE_NO, key: 'mode', val: 'int'}` at the beginning of all drill files (I'm assuming that Excellon has no concept of arcs). Then, if/when a slot command appears:

1. `{cmd: 'op', line: LINE_NO, key: 'move', val: {x: START_X, y: START_Y}}`
2. `{cmd: 'op', line: LINE_NO, key: 'int', val: {x: END_X, y: END_Y}}`

I'd appreciate a PR, otherwise I'll get to this shortly. Targeting v0.1.3 or v1.0.0, whichever comes sooner.

katyo, 2015-10-24:
I took a time machine and some voodoo to get excellon specification and convert it to pdf. So you can get it here:
https://gist.github.com/katyo/5692b935abc085b1037e/raw/32879038a55e1e7019902d8d073d6f5d6d2e45ff/excellon.pdf

mcous, 2015-10-24:
This is very helpful; thanks for your voodoo! Might have some time to code this up on Sunday; otherwise I'll try to get to it this coming week.

commit f8b6be90 (2015-11-16): Upgrade to 0.1.4 for drill slot support; fixes #3
closed 2015-11-16 by mcous