mcous/gerber-parser #8: Synchronous (not-streaming) parser API

katyo, 2015-11-27:
I want avoid of using readable-stream to parse gerber from string in browser. So I need a syncronous parser call. Is there some way to make it possible?

mcous, 2015-11-27:
Something like `var arrayOfCommands = gerberParser(aGerberString)` is not possible, because `gerber-parser` extends the the Node Transform stream. It must be used in an asynchronous manner. If you do not wish to use the streams interface, you could wrap it in some event listeners and use it with a callback. I'm going to assume you're passing an entire Gerber file as one string:

(Full disclosure, I just wrote this off the top of my head so it may not work completely properly)

``` javascript
var gerberParser = require('gerber-parser')

var gerberParserCallbackWrapper = function(gerberString, done) {
  var parser = gerberParser()
  var commands = []

  parser.on('warning', function(warning) {
    console.warn('warning at ' + warning.line + ' - ' + warning.message)
  })

  parser.once('error', function(error) {
    console.error('error - ' + error.message)
    parser.removeAllListerners('warning')
      .removeAllListeners('readable')
      .removeAllListeners('end')
    done(error)
  })
  
  parser.on('readable', function() {
    var data = parser.read()
    while (data) {
      commands.push(data)
      data = parser.read()
    }
  }

  parser.once('end', functino() {
    parser.removeAllListerners('warning')
      .removeAllListeners('readable')
      .removeAllListeners('error')

    done(null, commands)
  })

  parser.write(gerberString, 'utf8')
}
```

katyo, 2015-11-27:
Of course, I use something like this at current time:

```CoffeeScript
parseGerber = (str, cb)->
  parser = do createParser
  
  parser.on "error", (error)->
    cb error

  warnings = []
  parser.on "warning", (warning)-> warnings.push warning

  objects = []
  parser.on "data", (object)-> objects.push object

  parser.on "end", -> cb null, warnings, objects

  parser.end str
```

Ok, I will try to do things in async manner.

mcous, 2016-01-13:
I'm reopening this issue because I am considering adding a synchronous method to the parser. It's usage would look like:

``` javascript
var gerberParser = require('gerber-parser')

var parser = gerberParser()
var gerberString = getGerberStringSomehow()

parser.on('warning', function(warning) {
  console.warn('warning')
})

parser.on('error', function(error) {
  console.error('error')
})

var commandsArray = parser.transformSync(gerberString)
```

This would obviously be a blocking call, so if done in a browser it would freeze the UI etc. until finished unless it was used in a WebWorker. Would you still be interested in a synchronous interface?

katyo, 2016-02-09:
It's so good! Thank you.

commit c97713f3 (2016-02-28): Add synchronous parse method (fixes #8)
closed 2016-05-01 by mcous