mcous/gerber-parser #16: Explore new parsing mechanisms

mcous, 2017-05-26:
@kasbah after you linked me to that nearly article on gitter I noticed you launched a project using it. How do you like it / how hard was it to wrap your head around?

Currently we're entirely regex based, and I'm wondering if a more "standard" parsing mechanism like nearly could help us with stuff like #15

kasbah, 2017-05-26:
Nearly was fun to use and fairly easy to pick up, bearing in mind I had written a parser using parser combinators before which is a bit different but similarly "declarative" as a BNF style grammar. The [tutorial I linked to before](https://medium.com/@gajus/parsing-absolutely-anything-in-javascript-using-earley-algorithm-886edcc31e5e) is excellent as it suited my style of learning very well.

At this point I am not sure it will help you much with the immediate issues. The main advantage is that it's concise and clear but I think you give up some flexibility by defining it as grammar.  I am not sure if it's worth the effort at this stage. Maybe have a play on https://omrelli.ug/nearley-playground/

Other parser generators such as PEG.js are worth considering as well. Nearley's advantage is that you can define recursive grammars over PEG but you can almost always express your recursive grammar in a non recursive way as far as I understand. 

A quick search also led me to this EBNF definition of gerber: https://gist.github.com/corecode/58606f3178c4c301627a

mcous, 2017-05-26:
Do you feel like it could help with on the fly grammar selection? At the moment, the drill / gerber selection detection is pretty brute forced and not very nuanced.

kasbah, 2017-05-26:
It could make it less nuanced, sure. The input either matches the defined grammar or not (of either gerber _or_ drill). Has the brute force detection ever been an issue though?

mcous, 2017-05-26:
Haha no, it's always worked, it just makes me sad

kasbah, 2017-05-26:
Yeah, I would say defining a grammar is the _right_ way to do it and gives you that nice "it's all formally defined" feeling but at this point I think you have a really good parser already. Using an imperative style gives you more flexibility I think at the expense of having to read more code when working on it and having to hunt down where things are actually defined.

kasbah, 2017-05-26:
This is kind of a separate discussion but still fits the title. I just noticed that this is not actually a streaming parser even though it uses the `Readable`/ `Transform` Stream interface. That makes it kind of easier to do on the fly changing of coordinate format as we need to if we implement #15 but is there a plan to have an actual streaming output format i.e. potentially fire multiple `'data'` events and have the consumer make use of that?

mcous, 2019-09-19:
Exploration happening in the [lexer branch of tracespace/tracespace](https://github.com/tracespace/tracespace/tree/lexer). See WIP [packages/parser](https://github.com/tracespace/tracespace/tree/lexer/packages/parser)

closed 2019-09-19 by mcous