mcous/gerber-parser #19: EasyEDA gerbers not supported

henrikekblad, 2017-12-23:
One of the users tried to upload gerbers generated from EasyEDA. 

The parser pukes and complains about the coordinate system. 
```
Error: cannot parse coordinate with format undefined
    at parse (/home/oh/openhardware/node_modules/gerber-parser/lib/parse-coord.js:23:11)
    at parse (/home/oh/openhardware/node_modules/gerber-parser/lib/_parse-gerber.js:236:19)
```

[in wall scene controller power supply brd.zip](https://github.com/mcous/gerber-parser/files/1584367/in.wall.scene.controller.power.supply.brd.zip)

mcous, 2017-12-23:
Wow this is a frustrating bug. There's an extra space after the `*` in the `G04` comment line right before the `%FS` block.

```
G04 leading zeros omitted , absolute positions ,2 integer and 4 * 
%FSLAX24Y24*%
```

Since it strips newlines between blocks (stuff between `*`s and `*%`s), the parser is choking because it doesn't recognize ` %FSLAX24Y24` as the same thing as `%FSLAX24Y24`. Careless programming on my part; should be easy enough to fix

henrikekblad, 2017-12-23:
Wow, you're fast :) 

Happy X-mas btw.

mcous, 2017-12-23:
Thanks, you too! It's the only time of the year I have a little bit of free time ⌚️

closed 2017-12-23 by mcous

mcous, 2017-12-23:
Fix released as `v1.0.7`

henrikekblad, 2017-12-23:
Thanks! 
Can I expect you rolling this into gerber-to-svg/pcb-stackup also?

mcous, 2017-12-23:
It should show up in new installs of dependent tools thanks to the magic of semver. You may need to update lockfiles if you use them. If that's too much, though, ping me on gitter and I can probably find some time to push out some patch releases

henrikekblad, 2017-12-24:
Worked like a charm.

PR #20 (2017-12-23): (fix) Trim whitespace of parsed blocks
This PR updates the `get-next-block` function to trim the starting and ending whitespace of blocks. Extra whitespace can cause match failures downstream.

Fixes #19