mcous/gerber-parser #25: Loads of unrecognized blocks in the KiCad gerber

kasbah, 2018-06-06:
This is from https://kitspace.org/boards/github.com/emard/ulx3s/

gerber: [ulx3s-F.Cu.gtl](https://cloud.monostable.co.uk/ulx3s-F.Cu.gtl)

output of gerber-to-svg:

```
warning at line 56 - block "ADD24C,0.350000" was not recognized and was ignored
warning at line 59 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 59 - block "ADD25R,2.800000X2.000000" was not recognized and was ignored
warning at line 62 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 62 - block "ADD26R,2.800000X2.200000" was not recognized and was ignored
warning at line 65 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 65 - block "ADD27R,2.800000X2.800000" was not recognized and was ignored
warning at line 68 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 68 - block "ADD28R,0.700000X1.500000" was not recognized and was ignored
warning at line 71 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 71 - block "ADD29R,1.450000X0.900000" was not recognized and was ignored
warning at line 74 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 74 - block "ADD30R,1.450000X2.000000" was not recognized and was ignored
warning at line 77 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 77 - block "ADD31R,2.200000X1.800000" was not recognized and was ignored
warning at line 80 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 80 - block "ADD32R,0.560000X0.900000" was not recognized and was ignored
warning at line 83 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 83 - block "ADD33R,1.000000X0.670000" was not recognized and was ignored
warning at line 86 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 86 - block "ADD34R,1.500000X0.970000" was not recognized and was ignored
warning at line 89 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 89 - block "ADD35R,0.300000X1.900000" was not recognized and was ignored
warning at line 92 - block "%G04 #@! TD*G04 #@! TA.AperFunction,ComponentPad" was not recognized and was ignored
warning at line 92 - block "ADD36C,2.000000" was not recognized and was ignored
warning at line 95 - block "%G04 #@! TD*G04 #@! TA.AperFunction,SMDPad,CuDef" was not recognized and was ignored
warning at line 95 - block "ADD37R,0.845000X1.000000" was not recognized and was ignored
warning at line 98 - block "%G04 #@! TD*G04 #@! TA.AperFunction,ComponentPad" was not recognized and was ignored
warning at line 98 - block "ADD38R,1.700000X1.700000" was not recognized and was ignored
warning at line 101 - block "%G04 #@! TD*G04 #@! TA.AperFunction,ComponentPad" was not recognized and was ignored
warning at line 101 - block "ADD39O,1.700000X1.700000" was not recognized and was ignored
warning at line 104 - block "%G04 #@! TD*G04 #@! TA.AperFunction,ViaPad" was not recognized and was ignored
warning at line 104 - block "ADD40C,0.400000" was not recognized and was ignored
warning at line 107 - block "%G04 #@! TD*G04 #@! TA.AperFunction,ViaPad" was not recognized and was ignored
warning at line 107 - block "ADD41C,1.000000" was not recognized and was ignored
warning at line 110 - block "%G04 #@! TD*G04 #@! TA.AperFunction,ViaPad" was not recognized and was ignored
warning at line 110 - block "ADD42C,0.700000" was not recognized and was ignored
warning at line 113 - block "%G04 #@! TD*G04 #@! TA.AperFunction,ViaPad" was not recognized and was ignored
warning at line 113 - block "ADD43C,0.454000" was not recognized and was ignored
warning at line 116 - block "%G04 #@! TD*G04 #@! TA.AperFunction,ViaPad" was not recognized and was ignored
warning at line 116 - block "ADD44C,0.600000" was not recognized and was ignored
warning at line 119 - block "%G04 #@! TD*G04 #@! TA.AperFunction,ViaPad" was not recognized and was ignored
warning at line 119 - block "ADD45C,0.800000" was not recognized and was ignored
warning at line 122 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 122 - block "ADD46C,1.000000" was not recognized and was ignored
warning at line 125 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 125 - block "ADD47C,0.700000" was not recognized and was ignored
warning at line 128 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 128 - block "ADD48C,0.300000" was not recognized and was ignored
warning at line 131 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 131 - block "ADD49C,0.500000" was not recognized and was ignored
warning at line 134 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 134 - block "ADD50C,0.600000" was not recognized and was ignored
warning at line 137 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 137 - block "ADD51C,0.190000" was not recognized and was ignored
warning at line 140 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 140 - block "ADD52C,0.800000" was not recognized and was ignored
warning at line 143 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 143 - block "ADD53C,0.200000" was not recognized and was ignored
warning at line 146 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 146 - block "ADD54C,0.127000" was not recognized and was ignored
warning at line 149 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 149 - block "ADD55C,0.180000" was not recognized and was ignored
warning at line 152 - block "%G04 #@! TD*G04 #@! TA.AperFunction,Conductor" was not recognized and was ignored
warning at line 152 - block "ADD56C,0.254000" was not recognized and was ignored
```

mcous, 2018-06-06:
Appears to be choking on the `%` inside a comment block on line 56. This is against Gerber spec, so I'm not sure what to do.

Gerber spec section 3.6.6: 

> Strings are made up of all valid characters except the reserved characters CR, LF, ‘%’ and ‘*’.

Gerber spec section 4.14:

> The syntax for G04 is as follows.
> `<G04 command> = G04<Comment content>*`
> The `<Comment content>` must follow the syntax for strings in section 3.6.6. 

We could modify the parser to ignore `%` in comments and throw a warning? That doesn't seem too dangerous

kasbah, 2018-06-06:
I think it would be safe to ignore the `%` and throw a warning like you say. I'll file a bug with with KiCad as well.

kasbah, 2018-06-06:
Does seem to be a stray `%` and maybe not generated by KiCad? @emard did you do something that could have added the extra `%` in the F.Cu gerber?

kasbah, 2018-06-06:
Re-plotting this wth a nightly KiCad seems to replace most `G04` comments with `%` ones and doesn't have a stray `%`. 

```diff 
 G04 Gerber Fmt 4.6, Leading zero omitted, Abs format (unit mm)*
-G04 Created by KiCad (PCBNEW 5.0.0-rc2+dfsg1-2) date Wed Jun  6 08:38:01 2018*
+G04 Created by KiCad (PCBNEW 5.0.0-rc2-dev-unknown-44bf92a~82~ubuntu16.04.1) date Wed Jun  6 18:00:25 2018*
 %MOMM*%
...
%ADD17R,3.500000X3.300000*%
-G04 #@! TD*
-G04 #@! %TA.AperFunction,BGAPad,CuDef*
+%TD*%
+%TA.AperFunction,BGAPad,CuDef*%
 %ADD18C,9.000000*%
```

kasbah, 2018-06-06:
I guess you can't use it on Debian but I use this PPA on Ubuntu: https://launchpad.net/~js-reynaud/+archive/ubuntu/kicad-dev-nightly. It lets me run kicad-4 and nightly side by side. There is another one which replaces kicad-4 with the nightly.

emard, 2018-06-06:
If I can't install it on my debian, can I just
sed-edit trailing % from gerber files?

kasbah, 2018-06-06:
It's not the trailing % that are a problem but the one in the middle of a
comment block on line 56. I would just re-generate the gerbers and have a
look with http://viewer.tracespace.io if they work. Also, Debian unstable
will probably update the kicad-5 version soon enough?

emard, 2018-06-06:
Ups this one is on line 56
IG04 #@! %TA.AperFunction,BGAPad,CuDef*
If I sed-remove this % let's see what will be then.
I readdly hope for debian to update this sooon..

kasbah, 2018-06-07:
> issuecomment-395152932

You can also uncheck the X2 output in the KiCad plot dialog unless you need
it.

emard, 2018-06-07:
X2 was already unchecked, still %-junk generated